ut-robotics/picr22-team-teletupsud #53: Camera FOV cone top not parallel to ground

Akustav, 2022-11-30:
This might make it hard to see what is happening at a distance. Baskets might be harder to see. Did you consider it?

[image]

KaarelJ98, 2022-12-04:
Adjusted the camera mount angle by a few degrees. It's better now.

maidlam, 2022-12-04:
The FOV cone has a vertical angle of 40 degrees.
Actual vertical angle for the sensor is 42 degrees.

The actual view will be slighty above the ground parllel.
@Akustav would this cause any major issues?

[image]