ut-robotics/picr22-team-teletupsud #54: No mounting holes for thrower

Akustav, 2022-11-30:
How to you plan to fasten the thrower?

[image]

KaarelJ98, 2022-12-02:
I was thinking of fastening it from the top with 2 screws, since it can't move anyway because it is between top and bottom plate. Top and bottom plate will be fixed in place with screws in different places.

maidlam, 2022-12-04:
Adding bolts to the bottom might be a good idea.

During one test competition, the bottom plate of a robot came off from around a motor mount (no bolts were used). The robot didn't behave as expected then.

The bottom plate coming off from around the thrower could also cause issues.

KaarelJ98, 2022-12-04:
No problem, I'll add the holes for screws. Easy fix.